kevinakasam/ERCF_Filament_Cutter #2: Servo Hinge missing clearance

mobilemcclintic, 2023-11-06:
The servo hinge piece on B5 is missing a clearance cutout for the slide bearing stop(?).  This forces one side of the servo hinge to bend around it.
[image]

juliusjj25, 2023-11-20:
Ran into this same issue, created a part with clearance. 
[Servo_Hinge_Mod.zip](https://github.com/kevinakasam/ERCF_Filament_Cutter/files/13417989/Servo_Hinge_Mod.zip)

BioCookieYT, 2023-12-24:
Hi,
sorry for the late response and thanks for the feedback, I already opened a branch for this issue, going to work on a solution for this in a bit.

BioCookieYT, 2023-12-24:
Fixed with commit [58bf14c](https://github.com/kevinakasam/ERCF_Filament_Cutter/commit/58bf14c38e6058d93c8e38f4ddca13f30d7998c8) 
[image]

closed 2023-12-24 by BioCookieYT