kevinakasam/ERCF_Filament_Cutter #5: Work but...

Tutankamon00, 2023-12-07:
It Work, but :
1: I think it is not reliable, because plastic hole hold the cutter arm break with time
2: Movement and mechanical assembly of cutter arm is not stable, it shake during operation
3: Integration in MMU is not very good, because cutter is activated only during change filament in normal print, is best to activate it at every change of filament even not in print

BioCookieYT, 2023-12-24:
Hi, 
thanks for your feedback, be aware this project is far away from a final state, we are actively working on it on our discord (under the projects section). To answer your points:
1. Yes, we are aware of this issue and already working on this. We already released a new [servo arm](https://github.com/kevinakasam/ERCF_Filament_Cutter/blob/main/STL(Beta)/ERCF%20V1.0/Beta5/5.2/Servo_Arm.stl) which uses a different mounting system. Still doing some testing to find out what works the best. 
2. Could you share a video of this please? Most of our testers got it working smoothly. In the background @kevinakasam is working on a version which uses bearings, so I think it should be solved with that. 
3. The macros aren't perfect and are far away from being finished. We already had contact with the Happy-Hare developer to improve it and find solutions to our problems.

Tutankamon00, 2023-12-24:
>     2. Could you share a video of this please? Most of our testers got it working smoothly. In the background @kevinakasam is working on a version which uses bearings, so I think it should be solved with that.
> 
 Hi, i'am unable to make video because i have modified a little bit the gear, I don't know how to use fusion well, I did a bit of cutting and pasting to try to reduce the distance between the teeth of the two gears, this way it is more stable, I also enlarged the part on the encoder where the hole is for the servo arm, since the original one broke at that point, however I've been using it successfully for about 10-15 days, let's hope it lasts

closed 2024-07-25 by BioCookieYT