kevinakasam/ERCF_Filament_Cutter #6: Cutter Arm mod

Tutankamon00, 2023-12-26:
Hi, i suggest to modify the cutter arm like this, to incorporate the blade...
[image]

kevinakasam, 2023-12-26:
How would it cut the filament if the blade part that cuts the filament is cut out/missing 🤔

Tutankamon00, 2023-12-26:
> How would it cut the filament if the blade part that cuts the filament is cut out/missing 🤔

i using it from 10 days and work, the blade part is not missing: 
[image]

cedbossneo, 2024-01-02:
We should also incorporate a clearance for the tube, sometimes the filament blocks on the borders of it

sanpedrox2, 2024-01-31:
Stl? 😘

Tutankamon00, 2024-02-02:
> Stl? 😘

[cutterarm.zip](https://github.com/kevinakasam/ERCF_Filament_Cutter/files/14145081/cutterarm.zip)

sanpedrox2, 2024-02-02:
Nice!
Thanks mate :)

Tutankamon00, 2024-02-03:
> We should also incorporate a clearance for the tube, sometimes the filament blocks on the borders of it

Hi, please show your mod thx :)

BioCookieYT, 2024-02-15:
Thanks for the suggestion, your mod found the way in the new Beta6. Still have to update the Readme a bit and give some credits :p

closed 2024-02-15 by BioCookieYT