kevinakasam/ERCF_Filament_Cutter #7: Filament stuck

tutankino, 2024-01-05:
Hi, some time if Blade not cut the filament go down the Blade, because Blade arm with force go up , It Need a more robust design for cutter arm movement
[image]

BioCookieYT, 2024-01-22:
Hi!
Sorry for the late answer, we are already working on a more robust design on the discord server. KevinAkaSam had the time to try out a bearing version of the cutter, theoretically it would work better but we have to find a better way to mount the bearing. Even with the tightest press fit the bearing slides out after some time.
Could I ask you what Filament you tried cutting? Softer materials should work with the current design.

Tutankamon00, 2024-01-22:
I use pla+ sunlu and normal pla, with pla+ less problem, is bit soft compared to normal pla

BioCookieYT, 2024-02-15:
Hi again,
I reworked the cutter design a bit, could you please try out the new Beta6 and see how it goes now?
Thanks

Tutankamon00, 2024-02-15:
> Hi again, I reworked the cutter design a bit, could you please try out the new Beta6 and see how it goes now? Thanks

New Cutter Arm and Servo Arm for beta6 are compatible with beta5? i have ercf V1 with binky

closed 2024-07-25 by BioCookieYT