kevinakasam/ERCF_Filament_Cutter #8: Cutter_Arm collides with Servo_Mount

24Tigger24, 2024-01-06:
Hi, the Cutter_Arm collides with the Servo_Mount in the V2.0  
[image]

BioCookieYT, 2024-01-22:
Hi!
Sorry for the late answer and thanks for reporting this, I will work on a solution ASAP. It's a bit difficult time wise at the moment.

BioCookieYT, 2024-02-15:
Please try the new Beta, it should be fixed now. Still have to update the Readme a bit and give some credits :p

iamnotahippy, 2024-02-15:
I have printed the beta6 stls, and while I don't have my cutter wired up yet, the cutter arm no longer collides with the mount, and the gears no longer climb up one another when they encounter the resistance of cutting the filament. Also, I appreciate tapering the hole of the filament path through the cutter arm. I had to make the same modification to mine earlier to prevent jamming. Excellent work all the way around!

closed 2024-02-15 by BioCookieYT