kevinakasam/ERCF_Filament_Cutter #9: Macro CUTTER_ACTION called recursively

pmyrie, 2024-01-22:
hi, after updating happy-hare to 2.4 every time theh cutter action is call i get,

An issue with the MMU has been detected whilst out of a print
Reason: Error running _MMU_POST_UNLOAD: Macro CUTTER_ACTION called recursively.

BioCookieYT, 2024-01-24:
Hi, 
gonna find a solution ASAP. Please use HHv2.3 if possible

Raabi91, 2024-01-25:
> hi, after updating happy-hare to 2.4 every time theh cutter action is call i get,
> 
> An issue with the MMU has been detected whilst out of a print Reason: Error running _MMU_POST_UNLOAD: Macro CUTTER_ACTION called recursively.

please upload you klipper log

pmyrie, 2024-01-25:
Hi sorry for the delay.

[klippy.log](https://github.com/kevinakasam/ERCF_Filament_Cutter/files/14058736/klippy.log)

Tutankamon00, 2024-01-28:
Any news? because this i 'am unable to use ercf, it go in pause and not resume 
[image]

moggieuk, 2024-01-28:
Kevin,  the issue is that `MMU_EJECT` should not be called from within the `_MMU_POST_UNLOAD` macro because that will cause the recursive callback (yes, it just happened to work previously but should not have. lol).

The fix is to replace `MMU_EJECT` with `_MMU_STEP_UNLOAD_GATE`

So that the individual step of unloading the gate is called, not the whole unload sequence.

Tutankamon00, 2024-01-28:
I tried the Moggieuk solution and work perfect, thank you :)

BioCookieYT, 2024-01-28:
> Kevin, the issue is that `MMU_EJECT` should not be called from within the `_MMU_POST_UNLOAD` macro because that will cause the recursive callback (yes, it just happened to work previously but should not have. lol).
> 
> The fix is to replace `MMU_EJECT` with `_MMU_STEP_UNLOAD_GATE`
> 
> So that the individual step of unloading the gate is called, not the whole unload sequence.
> 
> So I'd recommend:
> 
> ```
> # New Cutter Action Macro - thanks to xF4m3
> [gcode_macro CUTTER_Action]
> description: Set Servo in the close position. Doesn't work properly yet.
> gcode:
>     {% set cutvar = printer["gcode_macro _CUT_VAR"] %}
>     
>     CUTTER_OPEN
>     MMU_TEST_MOVE MOVE={cutvar.feed_length + cutvar.cut_length}
>     {% for i in range(cutvar.cut_amount) %}
>       CUTTER_CLOSE
>       CUTTER_OPEN
>     {% endfor %}
>     MMU_TEST_MOVE MOVE=-1
>     CUTTER_CLOSE
>     
>    _MMU_STEP_SET_FILAMENT STATE=1 
>    _MMU_STEP_UNLOAD_GATE
>    
>     M400
>     G4 P0
>     ```
> ```

Thank you so much for helping out! I update the Readme in a bit.

closed 2024-01-28 by BioCookieYT