kevinakasam/ERCF_Filament_Cutter #11: Support for ERCF V1.1

24Tigger24, 2024-02-15:
According to my understanding, with Beta6 ERCF V1 is no longer supported because the new Binky Encoder no longer has a magnet to open the gates. So I added the magnets back (and 0.2mm clearance between the filament blocks and the encoder) to the 623 Bearing Version.
[image]
[V1.1 Binky Encoder for 623 Bearing.zip](https://github.com/kevinakasam/ERCF_Filament_Cutter/files/14300721/V1.1.Binky.Encoder.for.623.Bearing.zip)

BioCookieYT, 2024-02-15:
Hi,
did you find the time to test your mod out? I would like to add it in some kind of usermods section.

24Tigger24, 2024-02-16:
I just built it but currently don't have any multifilament prints planned. If the V2.0 encoder works, then my mod also works. The only thing I changed is the magnet on the back, which I can confirm can open the V1.1 gates.

sanpedrox2, 2024-02-16:
so... if i have a v1 with binky, springy, sturdy bunny and triple decky.. i pretty much have a V2 minus some bling?
and beta 6 should work on it?

24Tigger24, 2024-02-16:
Maybe... I designed the Encoder to work with the default V1.1 Filament blocks

BioCookieYT, 2024-02-16:
> so... if i have a v1 with binky, springy, sturdy bunny and triple decky.. i pretty much have a V2 minus some bling? and beta 6 should work on it?

With all the mods it is basically already a V2, if you are still using the triple decky magnet gates try this mod, otherwise the normal Beta6 files should work just fine

sanpedrox2, 2024-02-16:
ok, sweet. just noticed i wrote the wrong block mod. i use thumper blocks :) 
so i guess beta6 will work just fine as is for me then

24Tigger24, 2024-02-17:
I added the V-Bearing Version of the Encoder and removed the clearance i needed
[V1.1 Binky Encoder for EREC.zip](https://github.com/kevinakasam/ERCF_Filament_Cutter/files/14320109/V1.1.Binky.Encoder.for.EREC.zip)

BioCookieYT, 2024-02-22:
Hi again,
Do you know how to do Pull Requests on GitHub? I made a little [Usermods](https://github.com/kevinakasam/ERCF_Filament_Cutter/tree/main/Usermods) section in the repo, similar to the VORON usermods. I would like to add some mods for EREC there ^^

24Tigger24, 2024-02-23:
I don't know completly understand what pull requests are but yes, do so.

ntchris, 2024-06-30:
I am using ERCF system V1.0, but with binky style optical decoder.
I tried the Beta6 with beta6 encoder case, it works well.
so maybe do not say "beta6 dropped support" for ERCF1, hopefully more people still using ERCF1 can use this useful cutter addon.