kevinakasam/ERCF_Filament_Cutter #12: Suggestion

Tutankamon00, 2024-02-15:
Hi Pease modify a little the design, make the teeth of servo arm widher, to maximize the contact surface with cutter arm, this increase the stability and the durability :)
[image]

24Tigger24, 2024-02-25:
I haven't tried it yet
[Wider Servo Arm.zip](https://github.com/kevinakasam/ERCF_Filament_Cutter/files/14396845/Wider.Servo.Arm.zip)

Tutankamon00, 2024-02-25:
> I haven't tried it yet [Wider Servo Arm.zip](https://github.com/kevinakasam/ERCF_Filament_Cutter/files/14396845/Wider.Servo.Arm.zip)

Nice thank you, i'm try to adapt for ercf 1.1 with binky, if is possibile at least only for cutter_arm and servo arm you can share .f3d with history, because this new teeth design is very difficulty to mache mod without history, thank you very much

Tutankamon00, 2024-02-25:
i checked the step of wider servo arm, but you have make it wider in the wrong direction 
[image]

24Tigger24, 2024-02-25:
the cutter arm in your CAD is in the wrong position. You have to move it down.

My v1.1 ERCF in CAD: 
[my ERCF v1.1.zip](https://github.com/kevinakasam/ERCF_Filament_Cutter/files/14397229/my.ERCF.v1.1.zip)
(.f3d-file is to large for github)

Tutankamon00, 2024-02-25:
> the cutter arm in your CAD is in the wrong position. You have to move it down.
> 
> My v1.1 ERCF in CAD: [my ERCF v1.1.zip](https://github.com/kevinakasam/ERCF_Filament_Cutter/files/14397229/my.ERCF.v1.1.zip) (.f3d-file is to large for github)

YES you are right, i have to move down, thank you very much

BioCookieYT, 2024-07-25:
See Beta7 ;)

closed 2024-07-25 by BioCookieYT