kevinakasam/ERCF_Filament_Cutter #20: Wiring Diagram?

spikeon, 2024-04-24:
Hey guys. 

I'm looking at the documentation and I can't tell where exactly I'm supposed to plug what in to what.  If you could put together a couple of sentences or a diagram of how to attach the electronics of this to the rest of the ECRFv2 system (or point me at the documentation I didn't see), I would greatly appreciate it. 

Thanks!

BioCookieYT, 2024-04-26:
Noted! Working on it

What board are you using? In case you are using the BTT MBB, I would just follow the ERCFV2 manual and use at the end a free pin of [STPx](https://github.com/bigtreetech/MMB/blob/master/Hardware/MMB%20CAN%20V1.0-Pin.jpg?raw=true) . According to our testers the board has plenty enough power to supply the cutter servo.

spikeon, 2024-04-26:
> Noted! Working on it
>
>
>
> What board are you using? In case you are using the BTT MBB, I would just follow the ERCFV2 manual and use at the end a free pin of [...]

Yes, I'm using the BTT MBB.  Does that mean I don't need this buck converter thingy I bought?

BioCookieYT, 2024-04-26:
Some people are using the BTT MBB 5V supply without any problem but I would recommend using still a buck converter for now at least, I will try to contact BTT and hear their thoughts on this topic. So connect it like this

Servo Data Pin -> Some free STPx pin 
Servo + -> Buck converter +
Servo - -> Buck converter  -
**DON'T FORGET TO ADJUST THE VOLTAGE OF THE CONVERTER BEFORE CONNECTING THE SERVO!** The needed voltage is on the spec sheet of the servo, usually around 5-7.5V

spikeon, 2024-04-26:
Yeah, I figured that I'd screw that up so I bought a buck converter that outputs 5v.

https://www.amazon.com/dp/B01NALDSJ0?ref=ppx_yo2ov_dt_b_product_details&th=1

I don't trust myself with "If you set this dial incorrectly you destroy expensive things"

spikeon, 2024-04-26:
Speaking of the buck converter, do you have a recommended place to put it?  I was thinking about modifying the housing of the ERCFv2 BTT MBB to be a bit thicker so I could mount the converter to the inside of it.  But maybe printing it it's own housing and then gluing it to the outside would be better.

ntchris, 2024-06-29:
agree 
should add at least one line of description to somewhere about how to connect the servo signal pin to ercf controller board.
so hopefully it should be too much work.
for example "for BTT MBB, follow the ERCFV2 manual and use one free pin of [STPx]" is a good direction
thanks.

BioCookieYT, 2024-06-30:
Sorry guys for the long waiting, Beta7 is way more work than expected :p 
I quickly added a short explanation of the wiring in the Readme. Beta7 will have a proper wiring guide in the documentation.

ntchris, 2024-07-01:
Thanks for the super quick update!
I just finished the build today and did some servo test and check the BTT schematics, have some interesting finding, hopefully this could help other people have similar problems or future document.

At the beginning the servo doesn't move at all,  I spent some time to investigate using multimeter, signal generator, oscilloscope and have some finding.

1 A standard RC motor like MG996R usually require 5V signal input, and it sometimes accept slightly lower voltage signal as low as 2.9V, your mileage vary a lot by the seller/manuafurer/batch etc.  Unfortunately for BTT MBB CANbus ERCF board, all STP port are directly connected to the MCU STM32, therefore, all port has only 3.3V (at the very most) signal level, and some port has lower like 2.6V only. measured by scope. 

2. Not all STPX port are made equal.  Some STPX pin has higher voltage signal output than others. 
So with the out of standard signal voltage, we are only on luck, it's just lucky that the servo accepts the lower out of spec low voltage signal and works. I measured a few ports using a oscilloscope, some has low voltage as low as 2.6V output signal (ie PB8) and the servo doesn't accept or move at all, really, it's too low.

3. My BTT board, the last STP11 (PB2) output 3.3V, and the servo works very well.   Your mileage may vary.  So it's a good idea to check a few more STP port to find a better one.

4. Double check BTT's pin diagram and get the latest version, some version has PIN error.

5. On the whole BTT ercf board, there is **ONLY One** servo port has signal output of 5V, that is the **selector servo**, because BTT spent extra cost to put a signal level translator/level shifter attached to it  74LVC1G17, because ERCF originally has only ONE servo, the selector push down servo. 

6. If we want to take luck out of the equitation, we may need to add a cheap level shifter (for signal) a long with a DC DC buck converter (for power). shouldn't be too difficult to connect

[image]

ntchris, 2024-07-01:
all stop pins are directly connected to stm32 gpio
[image]

BioCookieYT, 2024-07-20:
Thanks a lot for your analyses,
for now I leave the wiring guide like this. I know it isn't optimal but it will do the job for now. Thinking about a custom adapter board to shift the logic level for the future. 
[image]

f00d4tehg0dz, 2024-08-14:
Has anyone been successful in using the Easy BRD? If so do you have a diagram handy? I cannot for the life of me get the servo gear to rotate through the PA7 data/signal pin

Blendan1, 2024-08-15:
> Has anyone been successful in using the Easy BRD? If so do you have a diagram handy? I cannot for the life of me get the servo gear to rotate through the PA7 data/signal pin

I used the extra connector to connect the servo and got the lower from an external 12/24 to 5v converter.

[image]

My wiring is a bit of a mess, but it works

f00d4tehg0dz, 2024-08-15:
> > Has anyone been successful in using the Easy BRD? If so do you have a diagram handy? I cannot for the life of me get the servo gear to rotate through the PA7 data/signal pin
> 
> I used the extra connector to connect the servo and got the lower from an external 12/24 to 5v converter.
> 
> ![PXL_20240815_033319234.MP.jpg](https://private-user-images.githubusercontent.com/136379342/358112161-1cc96b0c-e82f-4a9a-9007-1cec6d7df73c.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjM3MTE0NjksIm5iZiI6MTcyMzcxMTE2OSwicGF0aCI6Ii8xMzYzNzkzNDIvMzU4MTEyMTYxLTFjYzk2YjBjLWU4MmYtNGE5YS05MDA3LTFjZWM2ZDdkZjczYy5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwODE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDgxNVQwODM5MjlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jYTZjZDY4OTdiOTJiMDFjODJmOGY3YmY5N2U4NjAzNzMzZjc4OGI2NDZmZjY3YWE3NGFmN2RjZGZhNzk1NWJiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.Bxfc8uQ1QfBTwRyRPqpdVZxFtIl35mVlwVC2fY8ZvGE)
> 
> My wiring is a bit of a mess, but it works

Thank you! 
It's odd, but mine has the same wiring and triangle labs board, yet mine won't spin the gear. (I've tried two boards now)  Is your pin the same with the same pulse width? 

`[mmu_servo cut_servo]
pin: mmu:PA7			# Extra Pin on the ERCF easy Board
maximum_servo_angle: 180	# Set this to 60 for a 60° Servo
minimum_pulse_width: 0.0005	# Adapt these for your servo
maximum_pulse_width: 0.0025	# Adapt these for your servo
`

Blendan1, 2024-08-15:
I'm currently not at home, so if it's fine I will get back to you on Sunday/Monday when I'm home again. I'm not sure what my config was again 😅

f00d4tehg0dz, 2024-08-27:
> I'm currently not at home, so if it's fine I will get back to you on Sunday/Monday when I'm home again. I'm not sure what my config was again 😅

Just checking in if you ever had a chance to look? I still cannot get mine to work! Crazy.

Blendan1, 2024-08-27:
@f00d4tehg0dz sorry I  forgot, here's my config:
```
##########################################################################################
# The servo hardware configuration. Change the values to your needs.
# 
[mmu_servo cut_servo]
pin: mmu:PA7			# Extra Pin on the ERCF easy Board
maximum_servo_angle: 180	# Set this to 60 for a 60° Servo
minimum_pulse_width: 0.0005	# Adapt these for your servo
maximum_pulse_width: 0.0025	# Adapt these for your servo
```
my wiring is as in the Picture.

Maybe your servo is bad, try using the selector serve (or a different known good servo), if that dosn't work on this pin with the right config then maybe you have tow bad boards.

Otherwise could you send a picture of your board where the servo is pluged in, just in case something is wrong there. any maybe a sanity check if your using the right wires.

Oh and have you checked if your 12v/24v to 5v converter is good?

Sorry if I'm saying stuff that is obvious here, but i just want to make sure nothing is missed.

f00d4tehg0dz, 2024-08-27:
> Oh and have you checked if your 12v/24v to 5v converter is good?

It's funny you mentioned that. I recently bought a new 12v to 5v converter with a display, but I couldn't get it to work correctly. It turns out I had my In/Out reversed. After correcting the issue, the motor is spinning when entering the cutting commands. 

Thanks so much for the sanity check here!