kevinakasam/ERCF_Filament_Cutter #22: Too much bending and filament not cut properly

Blendan1, 2024-05-18:
When it cuts the filament it cuts it 95% but then just bends the rest out of the way, if I manually hold the rest straight it cuts it with ease.
So my suggestion would be to add something to hold the filament in place or to push against. It doesn't have to be completely straight, just 40° off would work too.

Also the servo mount twists and flexes a lot while cutting (that may be because of the filament bending out of the way)

iamnotahippy, 2024-05-18:
Have you printed the newest EREC parts? They should address all of the issues you're having: the mounting frame is stiffer, and there is an m3 heatset included in the encoder housing to act as an anvil for the blade to cut against. You can find these in the Discord channel, as they're not yet posted on the github repo.

BioCookieYT, 2024-05-20:
Hi,
could you try out the new Encoder I just pushed? It includes the heatset mod which should provide better support for the filament. 

https://github.com/kevinakasam/ERCF_Filament_Cutter/tree/main/STL(Beta)/Encoder/V25

AlexKid007, 2025-01-19:
Instead of an M3 heat insert, have you thought about a M2.5 heat insert? The new beta is much better than the older one but I still have some bending with some filaments.

BioCookieYT, 2025-01-19:
That might be actually a good idea! I quickly modified the V-Encoder version for a M2.5 heatset insert. There might be some issues with the bowden tube since the heatset isn't really able anymore to hold it in place. I have **not** tested the version yet due to some time issues but I attached the files in case you would like to try it. 
[Files.zip](https://github.com/user-attachments/files/18470522/Files.zip)

[image]

[image]

AlexKid007, 2025-01-21:
I will give it a try

AlexKid007, 2025-04-03:
Sorry for late replay. I printed it and tried it out. Cuts much better now.