kevinakasam/ERCF_Filament_Cutter #24: Config issues in Happy Hare 2.51

TheBlackMambata, 2024-05-26:
I think I already set up everything ok but maybe I'm missing something about the filament position and park position relating to the blade psition (and its cutting macro) maybe...
when I print something everything is ok until the first filament change and I've got this messages in the console:

00:22
[T2] < En <<<<<<<<< [Ex << (*) << Nz] LOADED 0.0mm (e:54.2mm)
00:22
Attempting to recover filament position...
00:22
idle_timeout: Timeout set to 72000.00 s
00:22
After fixing, call RESUME to continue printing (MMU_UNLOCK to restore temperature)
00:22
MMU issue detected. Print will be paused
Reason: Unload sequence failed: Error running _MMU_CUT_TIP: The value 'mmu_gantry_servo' is not valid for SERVO.
Occured when changing tool: T2 > T4

So, when the printer finish the wipe tower lines with the first filament, it goes in a position (which I think it's the park position) and it initially begins to do the unloading - loading moves (like in the form tipping, but I disabled it with "form_tip_macro: _MMU_CUT_TIP" instead of _FORM_TIP), then it suddenly gives me those messages while staying in the parking position with the filament inside the heat-throat yet.
What am I missing?

iamnotahippy, 2024-05-26:
You should keep _MMU_FORM_TIP as your tip forming macro, and instead edit:

```variable_user_post_unload_extension : 'EREC_CUTTER_ACTION'	; Executed after default logic```

in your mmu_macro_vars.cfg

TheBlackMambata, 2024-05-26:
Ok I understood what was the problem... Of course I need tip forming to unload the filament from the extruder, so i changed to _MMU_FORM_TIP and then the variable_user_post_unload_extension : "EREC_CUTTER_ACTION" do the rest.

Sorry for bothering

Thank you very much for your amazing work guys

TheBlackMambata, 2024-05-26:
> You should keep _MMU_FORM_TIP as your tip forming macro, and instead edit:
> 
> `variable_user_post_unload_extension : 'EREC_CUTTER_ACTION' ; Executed after default logic`
> 
> in your mmu_macro_vars.cfg

Yeah thanks of course I needed it XD

TheBlackMambata, 2024-05-26:
Sorry if I bother,
I've got now only a minor issue and after resolve it I think I'm good to go:

I tuned every position values of toolhead sensor, extruder to nozzle etc (I had the ercf v.1.1 version so I had to customize everything manually) but the filament loading is kinda "too fast" and now I've got a huge blob during the filament change that stucks on my nozzle and then goes to the wipe tower...   I haven't got a blobifier yet. 
Any suggestion?
With the same values I didn't have such a huge blob in ERCF v.1.1 but I had everything set properly on the slicer too wherease now I'm trying to follow the guides and let the HH do everything

Motalw, 2024-10-02:
I had the same problem today (Unload sequence failed: Error running _MMU_CUT_TIP: The value 'mmu_gantry_servo' is not valid for SERVO), and I found out that you need to set the "variable_gantry_servo_enabled" to false then you can use the 	_MMU_CUT_TIP macro without problem.